openbci-archive/OpenBCI_NodeJS #15: Samples out of order

andyh616, 2016-01-03:
When streaming from the board, I noticed that some of the samples are coming in out of order. here is a plot of a couple seconds of data.  You can see that there is a systematic pattern to which samples are being presented out of order, and the out of order sample number looks to be increasing linearly.

[image]

and here's some example data. If you look after 62, 24 is presented a second time.  could be a duplicate, or a sample from another block of data

```
array([[ 0],
       [19],
       [20],
       [21],
       [22],
       [23],
       [ 1],
       [ 2],
       [ 3],
       [ 4],
       [ 5],
       [ 6],
       [ 7],
       [ 8],
       [ 9],
       [10],
       [11],
       [12],
       [13],
       [14],
       [15],
       [16],
       [17],
       [18],
       [19],
       [20],
       [21],
       [22],
       [23],
       [24],
       [25],
       [26],
       [27],
       [28],
       [29],
       [30],
       [31],
       [32],
       [33],
       [34],
       [35],
       [36],
       [37],
       [38],
       [39],
       [40],
       [41],
       [42],
       [43],
       [44],
       [45],
       [46],
       [47],
       [48],
       [49],
       [50],
       [51],
       [52],
       [53],
       [54],
       [55],
       [56],
       [57],
       [58],
       [59],
       [60],
       [61],
       [62],
       [24],
       [63],
       [64],
       [65],
       [66],
       [67],
       [68],
       [69],
       [70],
       [71],
       [72],
       [73],
       [74],
       [75],
       [76],
       [77],
       [78],
       [79],
       [80],
       [81],
       [82],
       [83],
       [84],
       [85],
       [86],
       [87],
       [88],
       [89],
       [90],
       [91],
       [92],
       [93]])
```

andyh616, 2016-01-04:
What seems to be happening is that the first 0-255 samples are being *stuck* into the data stream at later times, in order, which is why this plot looks like it has a linearly increasing function over many blocks of data.  I'm going to start up and collect data a few more times to see if this is a consistent pattern, or just some weird artifact that we don't have to worry about

pushtheworldllc, 2016-01-04:
Hmm. That is weird. Let me know what you find!

--------------------------------------------
 
AJ Keller
Founder & CEO
Push The World LLC

Founder & President
Push The Code

Work: (843) 471-0039 <tel:(843)%20471-0039>
Cell: (203) 733-4538 <tel:(203)%20733-4538>

Push The World LLC <http://www.pushtheworldllc.com/>
1303 Camp Rd, Charleston, SC <https://www.google.com/search?client=safari&hl=en&q=1303+Camp+Rd+Charleston+SC&spell=1&sa=X&ved=0CBkQvwUoAGoVChMIvYy9luKpxwIVRp-ACh3zCQcw>

Like Us On Facebook <https://m.facebook.com/pushtheworldllc>

Warning: The information transmitted is intended only for the person or entity to which it is addressed and may contain confidential and/or privileged material. Any review, retransmission, dissemination or other use of, or taking of any action in reliance upon, this information by persons or entities other than the intended recipient is prohibited. If you received this in error, please contact the sender and destroy this material.

pushtheworldllc, 2016-01-07:
Unable to duplicate, placed private property on each sample, see some sample output in the attached txt file.
[myOutput.txt](https://github.com/OpenBCI/openbci-js-sdk/files/81647/myOutput.txt)

andyh616, 2016-01-08:
I'm closing this because I can't replicate it.  Must've been some weirdness with starting and stopping the system a bunch.  We should be mindful of this though...it would be good to somehow build in a 'harder' reboot option so that there's no funny business upon restarting the device

closed 2016-01-08 by andyh616