openbci-archive/OpenBCI_NodeJS #17: Modules not importing correctly...

pushtheworldllc, 2016-01-14:
Several developers have complained that the module is not requiring properly. 
The following do not import in the entry file:
var k = require('./OpenBCIConstants');
var OpenBCISample = require('./OpenBCISample');
var serialPort = require('serialport');

pushtheworldllc, 2016-01-14:
Found this:
https://discuss.atom.io/t/cant-use-app-module-in-render-process/17326/3
Which leads me to thinking we might have to use this:
https://github.com/atom/electron/blob/master/docs/api/remote.md

Not sure just digging

pushtheworldllc, 2016-01-14:
Thinking this might be a bug with electron, closing here and opening over on OpenEXP

pushtheworldllc, 2016-01-16:
So when i use the module in my program, i find the problem to come from v8. The function that throws the error is called `Module._resolveFilename` and it takes two inputs, a `request` and a `parent`.
The problem is that v8 is looking for `"./openBCISample" in paths `"/Users/ajk/Thinker/client" versus in the node_module. 
[image]

teonlamont, 2016-01-17:
I'm sorry but I don't understand what the error is that is related to loading the constants.

pushtheworldllc, 2016-01-17:
I have three .js files in the project:

* openBCISample
* openBCIConstants
* openBCIBoard

All three are module.exports and all three are at the same level in the same directory.
In `openBCIBoard` I have to `require` at least `openBCISample` so:
```js
var openBCISample = require('./openBCISample');
var k = openBCISample.k;
```

However THIS issue comes from when v8 is trying to load `openbci-sdk` at runtime in say `OpenEXP` and when it get's to the line 
```js
var openBCISample = require('./openBCISample');
```
It throws the error cannot find module `./openBCISample` because there is no file called `./openBCISample` in the main project directory of OpenEXP.

teonlamont, 2016-01-17:
ah, got it. because it's not properly interpreting local imports

pushtheworldllc, 2016-01-17:
Yes!!! Need to figure out how to do that!

--------------------------------------------
 
AJ Keller
Founder & CEO
Push The World LLC
Work: (843) 471-0039
Cell: (203) 733-4538

Warning: The information transmitted is intended only for the person or entity to which it is addressed and may contain confidential and/or privileged material. Any review, retransmission, dissemination or other use of, or taking of any action in reliance upon, this information by persons or entities other than the intended recipient is prohibited. If you received this in error, please contact the sender and destroy this material.

pushtheworldllc, 2016-01-17:
thanks to @tashoecraft we are closing this!

closed 2016-01-17 by pushtheworldllc