openbci-archive/OpenBCI_NodeJS #23: startStream does not return promise

andyh616, 2016-01-25:
The documentation says this should return a promise but is just writes a value to the board.  Should we make this a promise or change the comments?  If it's possible that the command won't work occasionally, we should probably return a promise.

```
    /**
     * Purpose: Sends a start streaming command to the board.
     * @returns {Promise} indicating if the signal was able to be sent.
     * Note: You must have successfully connected to an OpenBCI board using the connect
     *           method. Just because the signal was able to be sent to the board, does not
     *           mean the board will start streaming.
     * Author: AJ Keller (@pushtheworldllc)
     */
    OpenBCIBoard.prototype.streamStart = function() {
        this.streaming = true;
        this._reset();
        return this.write(k.OBCIStreamStart);
    };

    /**
     * Purpose: Sends a stop streaming command to the board.
     * @returns {Promise} indicating if the signal was able to be sent.
     * Note: You must have successfully connected to an OpenBCI board using the connect
     *           method. Just because the signal was able to be sent to the board, does not
     *           mean the board stopped streaming.
     * Author: AJ Keller (@pushtheworldllc)
     */
    OpenBCIBoard.prototype.streamStop = function() {
        this.streaming = false;
        return this.write(k.OBCIStreamStop);
    };
```

andyh616, 2016-01-25:
stopStream doesn't return one either

pushtheworldllc, 2016-01-25:
`.write()` returns a promise, so i figured that if i return this.write() that would in turn return a promise once `.write()` resolves. Is that not true?

andyh616, 2016-01-25:
oooh, thats a good question, i'm not sure but I would guess it does. maybe shoecraft can help us on this one

pushtheworldllc, 2016-01-27:
@andyh616 did you investigate this further?

tashoecraft, 2016-01-27:
Looks fine to me. If this.write() is a promise, then returning it will create a Promise object, to resolve when it is done. Should be able to then chain off streamStart and stop

pushtheworldllc, 2016-01-27:
Ok thank you @tashoecraft, you the real MVP. I'm closing this issue!

closed 2016-01-27 by pushtheworldllc