openbci-archive/OpenBCI_NodeJS #25: simulator sample rate optionally parameter doesn't work

andyh616, 2016-02-17:
I think the issue is here.  when you make the new openbciSimulator object, the simulatorSampleRate never makes it into the object, so its just set to the default 250 
```
OpenBCIBoard.prototype.connect = function(portName) {
        this.connected = false;

        return new Promise((resolve,reject) => {
            // If we are simulating, set boardSerial to fake name
            var boardSerial;
            /* istanbul ignore else */
            if (this.options.simulate || portName === k.OBCISimulatorPortName) {
                this.options.simulate = true;
                if (this.options.verbose) console.log('using faux board ' + portName);
                boardSerial = new openBCISimulator.OpenBCISimulator(portName, {
                    verbose: this.options.verbose
                });
            } else {
                /* istanbul ignore if */
                if (this.options.verbose) console.log('using real board ' + portName);
                boardSerial = new serialPort.SerialPort(portName, {
                    baudRate: this.options.baudRate
                },(err) => {
                    if (err) reject(err);
                });
            }
```

andyh616, 2016-02-17:
when I manually set it, it does the job:

```
                boardSerial = new openBCISimulator.OpenBCISimulator(portName, {
                    verbose: this.options.verbose,
                    samplerate: 20
                });
```

pushtheworldllc, 2016-02-17:
ok good problem solving @andyh616 looks like an easy fix, let's keep this issue open and see if we can get this impedance testing working with this update too, i think so minor tweaks on that as well might need to happen.

pushtheworldllc, 2016-03-23:
Fixed. Will be solved in 0.2.1

closed 2016-03-23 by pushtheworldllc