openbci-archive/OpenBCI_NodeJS #28: impedance testing not operating properly

pushtheworldllc, 2016-02-24:
The values that come back from an electrode are too high. There is an issue.

pushtheworldllc, 2016-03-23:
Well @andrewheusser i wanted to add @teonbrooks here too. I need your help with this! So I have it mainly working, on my dev branch. However all I need to do is taken an FFT at 31.5Hz and use the intensity at that bin as the voltage. I would really love some help with this and this seems right up your guys wheel house. I can teach you how to do the automated testing to get it working and such too!

andrewheusser, 2016-03-24:
@pushtheworldllc : going in tomorrow?  lets take a look then

pushtheworldllc, 2016-03-24:
No unfortunately I am heading down to southern jersey for easter weekend but I will be online and can definitely have a google hangout!

andrewheusser, 2016-03-30:
@pushtheworldllc hey I need a little more context here...where do we want to add an FFT?

pushtheworldllc, 2016-03-30:
We want to use an FFT to get the amplitude of the channel voltage at 31.5Hz because that’s the frequency of the impedance test signal. Does that make sense?

AJ

pushtheworldllc, 2016-03-30:
Where we want to add it is in the calculation of impedances!

andrewheusser, 2016-03-30:
@pushtheworldllc i understand that, can you point me to the point in the code where it should be implemented?

pushtheworldllc, 2016-03-30:
https://github.com/pushtheworldllc/openbci-js-sdk/blob/dev/openBCISample.js#L215-L237

Please fork my branch if possible! There are a bunch of changes, getting ready for a 0.3

andrewheusser, 2016-03-30:
ok will do...hmmm, I'm not sure it makes sense to put it here since an FFT requires multiple time points to compute. How will computing an FFT help with the impedance estimates anyway?

pushtheworldllc, 2016-04-03:
See PR #32 for version 0.3.0 for major changes leading to the resolution of this bug. Need to test.

aj-ptw, 2016-12-05:
Reopening issue due to #131

aj-ptw, 2017-02-11:
Final going to close this soon

commit 95d6529e (2017-03-02): Fixes #28 - Impedance not working properly
closed 2017-03-02 by aj-ptw