openbci-archive/OpenBCI_NodeJS #31: Node 5 compatibility

pushtheworldllc, 2016-03-11:
The serialport dependency does not work on node 4

pushtheworldllc, 2016-03-30:
Looks like Serialport will be updating soon https://github.com/voodootikigod/node-serialport/issues/688

reconbot, 2016-04-03:
Serialport does work on node 4 btw

teonbrooks, 2016-04-03:
@reconbot I think we had a typo above, we've had it work with v4 but v5 of Node would cause the crash. are you saying that it is fine for v5?

reconbot, 2016-04-03:
We currently test in v0.10-5 on Mac, Linux, and windows for nodejs.
Electron apps can be made to work but don't seem to out of the box.

Node 6 and better Electron support are in the near future.

pushtheworldllc, 2016-04-03:
@reconbot what is your suggested fix for this issue `/Users/ajk/BCI/openbci-js-sdk/node_modules/serialport/build/Release/node-v47-darwin-x64/serialport.node'`? In my electron application i am using a script that looks like `"preinstall": "electron-rebuild -v 0.36.9 -p -f -w serialport -e node_modules/electron-prebuilt"` and that seems to solve this issue. Thanks!!

pushtheworldllc, 2016-04-09:
Fixed #32

closed 2016-04-09 by pushtheworldllc