openbci-archive/OpenBCI_NodeJS #38: NTP

teonbrooks, 2016-04-10:
Let's get this ish settled. Excited about it. Essentially, each data packet will have a time stamp with it. for instance with OpenEXP, when the computer presents, we will have its ntp timestamp for the screen. when we retrieve the data packets, we could modify the packet to have the trigger/tag, then it will be marked in the data. or we can just print a trigger file and do the data correspondence later off-line

pushtheworldllc, 2016-04-10:
yup yup good thoughts! I'm at OpenBCI three days this week so by Friday should have this firmware ready to be tested!

pushtheworldllc, 2016-05-09:
just wrapping up automated testing and documentation. this is working though.

teonbrooks, 2016-05-09:
working?!? :) I am so ready to test this. we will need to have a similar ntp call for openexp, that way, we can write out an presentation event file in NTP time, so it would make it easier for analysis.

pushtheworldllc, 2016-05-09:
well the module already syncs up with an ntp server and each sample will now have a  time stamp value with it, so you should implement in OpenEXP with a call to `.sntpNow()` when ever you present a stimulus or something i believe

pushtheworldllc, 2016-05-09:
@teonbrooks I'm really glad you're excited for this new feature! looking forward to getting this merged and being the only SDK that can do time syncing with the openbci :D

teonbrooks, 2019-04-10:
closing because @andrewjaykeller has added NTP to the module

closed 2019-04-10 by teonbrooks