openbci-archive/OpenBCI_NodeJS #44: Simulate bio-like data

alexcastillo, 2016-04-19:
Is there a way the simulated data is more bio-like?

pushtheworldllc, 2016-04-19:
Record a lot of bio data, store and read from it? I would be cool with using it as long as it is stored in 24-bit signed integers or we convert the floats to 24-bit signed integers.

pushtheworldllc, 2016-04-19:
The main purpose of the simulator is for CI. So just want to make sure we are covered with the core functions.

pushtheworldllc, 2016-04-23:
Ok i think we can do this. I've been thinking about it for a couple days and talked with @teonbrooks  yesterday about it too. Seems like if we do a sum of sine waves that decrease in power as the frequency increases we should be able to simulate some brainwaves.

alexcastillo, 2016-04-23:
I like the idea!

teonbrooks, 2016-04-23:
we can make sure the time-frequency has the 1/f characteristic. that would help

pushtheworldllc, 2016-04-25:
from a guy with 15+ years experience confirming teon: "1/f + a little bump around 10 Hz and you have a pretty realistic power spectrum" 

@teonbrooks do you want to do this pr?

teonbrooks, 2016-04-25:
yeah, I would like to do that.

pushtheworldllc, 2016-05-01:
This issue is resolved with #56

pushtheworldllc, 2016-05-01:
Will be in 0.3.4

closed 2016-05-01 by pushtheworldllc