openbci-archive/OpenBCI_NodeJS #46: Need to expose the scale factor value, dynamic for each channel

pushtheworldllc, 2016-04-24:


pushtheworldllc, 2016-04-24:
I am finding that it is useful to expose the current scale value to multiply the raw channel value with when sending data to cassandra.

teonbrooks, 2016-04-24:
so you just want the results from the A-to-D conversion then?

pushtheworldllc, 2016-04-24:
Well I'm only sending 24bits of data to Cassandra for each channel to minimize overhead and bandwidth, so I also must send once the value to multiple each 24 bit number by

teonbrooks, 2016-04-24:
yeah, that is what i was talking about a while ago. didn't you suggest having a way to toggle data output. you could add a param when it starts up analog or digital and in the data packet for analog the `scale = 1` and for digital, `scale =  scaleFactor`.

pushtheworldllc, 2016-04-24:
in a nutshell, there is more to it but yeah, we can now do other stuff with the stream packets to indicate to the computer-driver to do something different with it

teonbrooks, 2016-04-24:
oh, you mean to keep the packet compatible for impedance test as well as data streaming? because the scaleFactor is static scalar

pushtheworldllc, 2016-04-24:
yea, the four immediate cases are, and keep in mind this only really pertains to sending different types of data packets:

1. 8 channels with 3 accel
2. 8 channels but don't scale aux bytes
3. User defined stream packet (you are doing on board DSP and want to send other info) 
4. Time Stamped packet accel (8 chan, 4bytes for time 2bytes for accel) 
5. Time Stamped packet aux (8 chan, 4bytes for time 2bytes for aux)