openbci-archive/OpenBCI_NodeJS #63: ENH: Add Mac/Windows large serialport as option to simulator

pushtheworldllc, 2016-06-13:
The Mac and Windows FTDI virtual comm port drivers have big buffers of 4096 which create a fragmented data pulse, there must also be a time component, like every 500 ms it get's cleared, this should be looked into and mimicked in the simulator to provide a better testing environment for graphics developers.

baffo32, 2016-10-19:
Do you have any more information about how these packets appear?  Do you mean that data is buffered for ~500ms and then drained in one huge go, with many packets in one buffer?

Alternatively, could you throw a `console.log('_processBytes: ' + JSON.stringify('' + data));` into _processBytes and paste some sample output here?

aj-ptw, 2016-10-19:
It is entirely based on your virtual comm port driver. You have linux, from what i remember so check out your FTDI input buffer size and latency timer. These both work together to determine how often your serial port is dumped.

aj-ptw, 2016-10-21:
Closed with version `1.4.0`

commit 5b702683 (2016-10-19): Add simulation for buffer size and latency time (fixes #63)

Also minor changes to address comments on PR #100
closed 2016-10-21 by aj-ptw