openbci-archive/OpenBCI_NodeJS #66: issue with board initialization

andrewheusser, 2016-06-24:
```
var ourBoard = require('openbci-sdk').OpenBCIBoard({
...     verbose: true
... });
```
returns
```
TypeError: this.numberOfChannels is not a function
    at OpenBCIFactory.OpenBCIBoard (/Users/daniel.gulyas/dev/openbci-narcolepsy/node_modules/openbci-sdk/openBCIBoard.js:97:69)
    at repl:1:39
    at REPLServer.defaultEval (repl.js:262:27)
    at bound (domain.js:287:14)
    at REPLServer.runBound [as eval] (domain.js:300:12)
    at REPLServer.<anonymous> (repl.js:431:12)
    at emitOne (events.js:82:20)
    at REPLServer.emit (events.js:169:7)
    at REPLServer.Interface._onLine (readline.js:211:10)
    at REPLServer.Interface._line (readline.js:550:8)
```
but
```
        var OpenBCIBoard = require('openbci-sdk');
        var ourBoard = new OpenBCIBoard.OpenBCIBoard();
```
works fine!

pushtheworldllc, 2016-06-24:
Which version are you using?
Is this error on initialization?

andrewheusser, 2016-06-24:
this was someone else's error, but I've run into this as well.  they are using openbci-sdk 0.3.7 and node 4.4.6.  Yes, the error is on init

andrewheusser, 2016-06-24:
i always use this form to initialize: 
```
        var OpenBCIBoard = require('openbci-sdk');
        var ourBoard = new OpenBCIBoard.OpenBCIBoard();
```

pushtheworldllc, 2016-06-24:
ok will investigate! Thanks for the report!

pushtheworldllc, 2016-06-24:
wow... epic fail on my part! The read me was wrong every where but a couple of them! Big thanks to "someone else" @andrewheusser Please checkout this PR and merge if it looks good! Thanks for the report and follow through. ~AJ

commit b36a7b34 (2016-06-24): Merge pull request #67 from pushtheworldllc/dev

Fix constructor in readme.md Closes #66
commit cfa8f4bd (2016-06-24): Fix constructor in readme.md Closes #66
closed 2016-06-24 by pushtheworldllc