openbci-archive/OpenBCI_NodeJS #75: Don't emit accelerometer array when 0

aj-ptw, 2016-08-19:
The OpenBCI firmware sends packets at 250Hz, however the accelerometer is only at 25Hz. This leads to accelerometer data only being sent every 10 packets. 

Proposal is to align standard with time synced packets in that an `auxData` `Buffer` 6 bytes long is emitted with every sample and if the data is greater than zero we will omit an `accelData` array with converted and scaled accelerometer values.

aj-ptw, 2016-08-19:
@haniawni this is a proposal from your request.

teonbrooks, 2016-09-06:
keeping it periodic will make it a lot easier if you are trying to interpolate this signal to match the sampling rate of the EEG data

aj-ptw, 2016-09-06:
so you're saying don't change it?

teonbrooks, 2016-09-06:
I guess I don't understand the problem so clearly. occasionally the data are zeros?

aj-ptw, 2016-09-06:
yeah at the data rate of 250Hz one would only get valid accel data every 10 samples. So it was confusing for @haniawni as to why some array's had data while others did not.

closed 2016-12-05 by aj-ptw