openbci-archive/OpenBCI_NodeJS #87: Time sync feature not working with firmware version 2

rlqmal, 2016-10-03:
I've tried to use time sync feature by running `timeSync.js` in `examples/timeSync`.

I already updated my OpenBCI HW to firmware version 2. However, `syncClocksFull()` function is not working.

``` sh
bash-3.2$  ls
node_modules package.json timeSync.js
bash-3.2$ node timeSync.js
connected
(node:89295) Warning: Possible EventEmitter memory leak detected. 11 synced listeners added. Use emitter.setMaxListeners() to increase limit
^C
```
Have any idea?

aj-ptw, 2016-10-03:
i just saw this same issue when using similar code in the example file this weekend!! this is a bug and i will try to get tot eh bottom of it. Without looking at anything, I have to imagine this is from an `on` event being added and not removed.

rlqmal, 2016-10-03:
Thank you for the super fast answer @aj-ptw !

Software reset command  (`ourBoard.usingVersionTwoFirmware();`) returns true, so I think this bug may come from firmware side. I'll report, if I found something.

aj-ptw, 2016-10-03:
This is most likely not from the firmware (i wrote this into the firmware). This is from the parser in this module not finding the time sync sent confirmation `,` sent back from the `Dongle`. If the time sync set is not picked up, then the `syncClocksFull` function will `reject` after one second. But I imagine if we are syncing multiple times a second and some of the syncs do not complete successfully, then we will install to many event emitters.

aj-ptw, 2016-10-03:
Which version of the firmware are you using? Can you port your output like I did below?
[timeSyncExampleOutput.txt](https://github.com/OpenBCI/OpenBCI_NodeJS/files/506488/timeSyncExampleOutput.txt)

You can do it with:
`node timeSync.js > timeSyncExampleOutput.txt`

rlqmal, 2016-10-04:
Code that I wrote is raw output. 

```sh
bash-3.2$  ls
node_modules package.json timeSync.js
bash-3.2$ node timeSync.js
connected
(node:89295) Warning: Possible EventEmitter memory leak detected. 11 synced listeners added. Use emitter.setMaxListeners() to increase limit
^C
```

It doesn't print lines like `timeOffset 1475505398028`. And I updated OpenBCI board with [v2.0.0-rc.6](https://github.com/aj-ptw/OpenBCI_32bit_Library/releases/tag/v2.0.0-rc.6) from your repository [OpenBCI_32bit_Library](https://github.com/aj-ptw/OpenBCI_32bit_Library).

aj-ptw, 2016-10-04:
Please use the radio code found here:
https://github.com/OpenBCI/OpenBCI_Radios/tree/2.0.0

And the library code found here:
https://github.com/OpenBCI/OpenBCI_32bit_Library/tree/2.0.0

aj-ptw, 2016-10-04:
Let's at least get you working and seeing those timeOffset prints and go from there. @rlqmal I am still seeing the possible event emitter code with working time sync so I know we should at least be able to get you going!

AJ

rlqmal, 2016-10-11:
Sorry for late response. I tried to upload radio code you suggested. Host radio code works fine. However, uploading radio device code is not working. 

Arduino reports while uploading RadioDevice32bit.ino
```
get
fail.......fail.......fail.......
```
I followed basic pin connection from [UPLOAD CODE TO OPENBCI RADIOS](http://docs.openbci.com/tutorials/03-Upload_Code_to_OpenBCI_Dongle).

Question: Do I have to update radio device also?

aj-ptw, 2016-10-11:
> Question: Do I have to update radio device also?
Yes for sure!

> uploading radio device code is not working.
You need to make sure you are holding the wires on firmly and the board is powered on. I find it helps to put the board top down, run the four wires through a header, and push firmly!

rlqmal, 2016-10-11:
I think my first OpenBCI (I have two boards.) is broken. Uploading radio firmware to the second OpenBCI was successful. 

However, I found an another error. While uploading the library code, it shows error in verify flash step. Have any idea? 
```
Programmer for Microchip PIC32 microcontrollers, Version 2.0.180
    Copyright: (C) 2011-2015 Serge Vakulenko
      Adapter: STK500v2 Bootloader
 Program area: 1d000000-1d1fffff
    Processor: Bootloader
 Flash memory: 2048 kbytes
  Boot memory: 80 kbytes
         Data: 74360 bytes
        Erase: done
Program flash: .....................................###################################### done
 Verify flash: .....................................
error at address 1D0000FC: file=9D001250, mem=9D00125C
``` 
Thanks!

aj-ptw, 2016-10-16:
Hi @rlqmal how is it going? How can I help?

rlqmal, 2016-10-18:
All my board is broken. Is there any way that I can reset my OpenBCI board?

aj-ptw, 2016-10-18:
Can you at least stream data from the Board?

closed 2016-11-01 by aj-ptw