openbci-archive/OpenBCI_NodeJS #91: Timeouts used to ensure commands are written rather than promise resolution

baffo32, 2016-10-16:
Throughout the code, timeouts are used to wait for commands to be written.  This could make maintainability more difficult if the command delay ever changes, and is nonintuitive for users of the library.

Ideally write functions should resolve a promise only when their write has completed, and this promise resolution should be used to know that the writing has completed.

A simpler approach could be to adjust `.disconnect` to wait until all writes have completed, but this would not solve the problem in as many cases.

aj-ptw, 2016-10-16:
Agreed! Version 2 of the firmware has no delays for 99% of all cases.

> Ideally write functions should resolve a promise only when their write has completed, and this promise resolution should be used to know that the writing has completed.
Any idea how to link the command with resolution of it's promise call?

baffo32, 2016-10-16:
Instead of lengthening an array of commands, one could have an array of objects which each entry contain the command, the resolve function, and the rejection function.
Alternatively one could have a new local function call made for each command, and use local variables.  Then these functions are stored rather than the command.

commit d06b6101 (2016-11-10): Resolve write promises when writes complete.  Fixes #91
closed 2016-11-16 by aj-ptw