openbci-archive/OpenBCI_NodeJS #97: Commands with responses resolve early

baffo32, 2016-10-17:
Commands that return data should not resolve until they can resolve with the return value.

[ ] implement a generic function such as `writeForEOT` that writes and then waits for the next EOT string to resolve with.  It will have to behave differently depending on the read state that precedes its physical write.
[ ] rewrite existing commands to use this function

Implementing this may involve reviewing and re-evaluating the current 'state-based' approach to reading data; or it may be straightforward to include.

aj-ptw, 2016-11-08:
@baffo32 any idea what we can do to merge this in? I'm looking to get `1.4.1` merged in ASAP, has a couple good bug fixes.

baffo32, 2016-11-09:
I haven't started this yet; I don't feel I have a good enough handle yet on all the responses that can come from the board.  I was really on a roll when I opened this issue, but less time has been available to me more recently.

aj-ptw, 2017-01-03:
This was completed.

closed 2017-01-03 by aj-ptw