openbci-archive/OpenBCI_NodeJS #99: Tests fail on real stock v1 board

baffo32, 2016-10-17:
Tested for the first time with a real board, since I was refactoring a little and wanted to ensure things still worked.  The SNTP failures are normal for me, but the others aren't.

This is a 32-bit board I received from OpenBCI in early-mid 2016.  I have never upgraded the firmware.

```
  1) #impedanceTesting "before all" hook:
     Error: timeout of 20000ms exceeded. Ensure the done() callback is being called in this test.
  

  2) openbci-sdk #boardTests #connect/disconnect/streamStart/streamStop gets the ready signal from the board and sends a stop streaming command before disconnecting:
     Error: timeout of 3000ms exceeded. Ensure the done() callback is being called in this test.
  

  3) openbci-sdk #boardTests #connect/disconnect/streamStart/streamStop rawDataPacket is emitted:
     Error: done() invoked with non-Error: Error [.streamStart()]: Already streaming
      at test/openBCIBoard-test.js:462:47

  4) openbci-sdk #boardTests #connect/disconnect/streamStart/streamStop rawDataPacket is emitted:
     Error: done() invoked with non-Error: Error [.streamStart()]: Already streaming
      at test/openBCIBoard-test.js:462:47

  5) openbci-sdk #boardTests #sdStart can start 14 seconds of logging with sd:
     Error: timeout of 3000ms exceeded. Ensure the done() callback is being called in this test.
  

  6) openbci-sdk #boardTests #sdStart "after each" hook for "can start 14 seconds of logging with sd":
     Error: done() called multiple times
      at Suite.<anonymous> (test/openBCIBoard-test.js:510:7)
      at Suite.<anonymous> (test/openBCIBoard-test.js:502:5)
      at Suite.<anonymous> (test/openBCIBoard-test.js:414:3)
      at Object.<anonymous> (test/openBCIBoard-test.js:17:1)
      at require (internal/module.js:12:17)
      at Array.forEach (native)
      at node.js:974:3

  7) openbci-sdk #boardTests #sdStart "after each" hook for "can start 14 seconds of logging with sd":
     Error: done() called multiple times
      at Suite.<anonymous> (test/openBCIBoard-test.js:510:7)
      at Suite.<anonymous> (test/openBCIBoard-test.js:502:5)
      at Suite.<anonymous> (test/openBCIBoard-test.js:414:3)
      at Object.<anonymous> (test/openBCIBoard-test.js:17:1)
      at require (internal/module.js:12:17)
      at Array.forEach (native)
      at node.js:974:3

  8) openbci-sdk #boardTests #sdStart "after all" hook:
     Error: done() invoked with non-Error: no board connected
      at test/openBCIBoard-test.js:520:25

  9) openbci-sdk #time should emit sntpTimeLock event after sycned with ntp server:
     Error: timeout of 2000ms exceeded. Ensure the done() callback is being called in this test.
  

  10) openbci-sdk #sntpStart should be able to start ntp server:
     Error: timeout of 2000ms exceeded. Ensure the done() callback is being called in this test.
  

  11) openbci-sdk #sntpStop "before all" hook:
     Error: timeout of 2000ms exceeded. Ensure the done() callback is being called in this test.
  

  12) openbci-sdk #radioTestsWithBoard "before all" hook:
     Error: Error Resource temporarily unavailable Cannot lock port
      at Error (native)

  13) openbci-sdk #hardwareValidation "before all" hook:
     Error: Error Resource temporarily unavailable Cannot lock port
      at Error (native)

  14) #daisy "before all" hook:
     Error: timeout of 4000ms exceeded. Ensure the done() callback is being called in this test.
  

  15) #syncWhileStreaming "before all" hook:
     Error: timeout of 4000ms exceeded. Ensure the done() callback is being called in this test.
  

  16) #syncErrors "before all" hook:
     Error: timeout of 4000ms exceeded. Ensure the done() callback is being called in this test.
```

aj-ptw, 2016-10-17:
If i had a known issues section i would put this here...

The tests worked at some point a while ago but at one point I lost it and have been meaning to get it working but ran out of time. I really would love to get this going.

aj-ptw, 2016-10-17:
I think a majority of tests should not use the real board, there needs to be a better separation of functional tests from unit tests.

aj-ptw, 2016-10-20:
> Detect this situation and instruct the user to powercycle their OpenBCI (and upgrade their firmware)

haha love it

aj-ptw, 2016-10-20:
I think you're doing great work. Reviewing as fast as I can! Thanks!