openbci-archive/OpenBCI_NodeJS #112: consider bluebird for Promise and enable developer warnings

baffo32, 2016-10-23:
Using bluebird may create more robust code: http://bluebirdjs.com/docs/features.html#debuggability-and-error-handling

baffo32, 2016-10-23:
bluebird is compatible with other promise libraries, but would create an additional dependency that users of the library would have to install; it could alternatively be used only for tests and the dependency saved, but that would deny developers of its additional features

aj-ptw, 2016-10-23:
Yea but it wouldn't result in a breaking change right

baffo32, 2016-10-25:
It would break obscure uses that don't limit themselves to the public promise API specification.

Right now I'm working on a change that uses bluebird only for testing.

baffo32, 2016-10-31:
I'll close this since it's in tests now.
I think it could moved to be used fully internally no problem.

closed 2016-10-31 by baffo32