openbci-archive/OpenBCI_NodeJS #114: Return an Observable instead of firing events

jspenc72, 2016-10-26:
Anyone interested in an RxJS Observable variant of this project?

https://github.com/Reactive-Extensions/RxJS/blob/master/doc/api/core/operators/start.md

aj-ptw, 2016-10-26:
Would that simplify the interface?

jspenc72, 2016-10-26:
Yes

jspenc72, 2016-10-26:
See this package as an example where an Observable is returned.
https://www.npmjs.com/package/robinhood-observer

jspenc72, 2016-10-26:
If there is a lot of interest from the community I would be happy to create this.

jspenc72, 2016-10-26:
```typescript
var OpenBCI = require('openbci-observer')     
var observer = OpenBCI(null).observeAll()  
 
var dataSubscription = observer
                    .map(tick => tick.results)                            
                    .distinct()                   //Only use distict results... 
                    .subscribe(x => {
                        //This block of code is executed each time an event would have been called.
                        console.log(x);
 
                    }, e => {
                        console.error(e)
                    }, () => console.log('data subscription disposed'));

//Unsubscribe to updates after 5 seconds. 
 
setTimeout(function(){
 subscription.dispose();  
}, 5000);

```

jspenc72, 2016-10-26:
Thoughts @aj-ptw ???

jspenc72, 2016-10-26:
Yes, see this package as an example. 

https://www.npmjs.com/package/robinhood-observer

aj-ptw, 2016-10-26:
Would we have to have a different repo or could we make this default?

There is a lot of interest from the community to simplify the number of functions needs to stream to one.

aj-ptw, 2016-10-26:
@tashoecraft you work with RxJS, good application for it? 
@alexcastillo thoughts? 
@baffo32 you have been helping a lot, what are your thoughts?

aj-ptw, 2016-10-26:
@jspenc72 I think this is super cool and would love it for you to contribute it, just checked out the Robin Hood wrapper, still I don't understand why it's in another repo.

baffo32, 2016-10-26:
@jspenc72 were you considering going 'under the hood' and changing how the library functions to use rxjs instead of promises (if that even makes sense) internally?  Perhaps it could be helpful to have another set of eyes reviewing and organizing the internal code.

I am working detecting runaway promises in tests.  So, I would be curious if this will still be reasonable to do with the addition of rxjs: to assert at particular points in the program that, globally, no tasks are still running.

Looks good to me, at first glance.

aj-ptw, 2016-10-26:
@baffo32 I think this flows really well post your #95 and all the clean up work you're doing. If we are able to connect and well disconnect without resetting, this could play really well with the wrapper @jspenc72 wants to build. The more I think about it, seems like it would almost be a requirement. Or the wrapper, just doesn't disconnect once connected.

jspenc72, 2016-10-26:
@aj-ptw , the robinhood-observer is in a different repo because the owner of the robinhood repo felt that observer were not needed and would be better built as a standalone project.

jspenc72, 2016-10-26:
You don't have to have a different repo. Yes I would likely "go under the hood"

jspenc72, 2016-10-26:
Ideally the observable would be tied directly to the incoming data stream from the device at the ealiest place possible. So if you have a stream coming in then the observable would be generated at when the stream is created and will be updated each time you receive a new packet from the stream.

aj-ptw, 2016-10-26:
Ok love it @jspenc72 

The current roadmap is:
### 1.4.0 - Connection - Mid November
#96 is going to and has some big changes from a developer standpoint with #83 merge conflicts could be big. @baffo32 has been working on that and it's looking great. 
Connect and disconnect from board with out stopping streaming.
### 1.4.1 - CLI - End of November
#50 is a command line interface for the board.
Try to have all tests passing with a real firmware `v1` board.
### 2.0.0 - Ganglion - December 2016
Looks like shipping is happening at the end of Novmeber from what I hear.
Requires creating an index entry file.
Requires moving core code to central file for boards to share.
Requires `noble` or node ble.
### 3.0.0 - Wifi - February 2017
Connect to the board(s) through the new wifi shield.
Enables faster sampling rate. (No more sending 125Hz with Daisy board, aiming for 1000Hz)

I have really tried to make auto testing a prioity so please feel free to ask any questions about writing or running tests on the code you contribute. Given that road map,

tashoecraft, 2016-10-27:
@aj-ptw I think making it Observable based is very logically and fitting.

jspenc72, 2016-10-28:
I reviewed the code and see that there are come cases where a promise is probably "ok".

Do you see any advantage in keeping any of the existing promise based methods?

jspenc72, 2016-10-28:
ie 

.connect()

.disconnect()

Etc...

jspenc72, 2016-10-28:
To create an observable to handle these might make that portion less friendly than a Simple

```js
.then().catch()
```

Interface

jspenc72, 2016-10-28:
Thoughts? @aj-ptw

aj-ptw, 2016-10-28:
@jspenc72 anything is on the table. We are going for simplification so if creating an observable would only make it worse... i would say that's probably not best. Is there a circumstance where users would _not_ want to use observables or even not be able to use observables? 

```
var dataSubscription = observer
                    .map(tick => tick.results)                            
                    .subscribe(x => {
                        //This block of code is executed each time an event would have been called.
                        console.log(x);  
                        // x would be the json representing the data received from the headset.

                    }, e => {
                        console.error(e)
                    }, () => console.log('data subscription disposed'));
```

This would be cool to get it down to just one line. Where you pass in the port name and boom, now you get events.

jspenc72, 2016-10-28:
Ill write a few lines of code tonight and see what I can do

jspenc72, 2016-10-28:
Yes, Well definitely keep promises around. Right tool for the right job you know.. 👌

baffo32, 2016-10-28:
One note: be sure to do new work on the 1.4.0 branch so merging is reasonable.  All the whitespace changed.

jspenc72, 2016-10-28:
I think it will help to abstract away the tedious stuff required to start streaming data by default with a one liner observable created as shown above. But allow the user to manually call the methods by passing a config object, possibly when they import the library.

jspenc72, 2016-10-28:
Will do @baffo32

alexcastillo, 2016-10-30:
+1 for Observables support using RxJS.
Let me know if you would to get some input from the RxJS team.

aj-ptw, 2016-12-05:
@jspenc72 any progress on this? Branch 2.0.0 is looking about good on my end.

jspenc72, 2017-02-20:
@aj-ptw sorry for the delayed response i have been playing with an RxJS implementation... and I think the best solution is to make a new repository as you previously mentioned... Although the raw data processing will remain similar the interface has become extremely different, doesn't resemble this project at all and is a complete rewrite.

I have created an sdk which would be a replacement that is written in TypeScript and will happily share it so we can get feedback from the community soon.

Cheers!

jspenc72, 2017-02-20:
FYI its ES6 and fully compatible with the latest version of node.

alexcastillo, 2017-03-24:
Alright, after meeting with @teonbrooks we decided that the best way to move forward with our project is to create a lib that abstracts all the common operations (filtering, buffering, windowing, FFT, etc). So, I took a first pass at implementing Reactive Extensions for OpenBCI.

Here's the project repo and package:
https://github.com/NeuroJS/openbci-rx
https://www.npmjs.com/package/openbci-rx

Do you guys mind helping by testing? 
Please feel free to log any issues in the issues page of the repo.

aj-ptw, 2017-03-24:
Wow awesome @alexcastillo closing this issue for now

closed 2017-03-24 by aj-ptw