openbci-archive/OpenBCI_NodeJS #115: Some samples dropped when received in a large clump.

baffo32, 2016-10-30:
This is resulting in occasional test timeouts, when samples are processed too infrequently to finish in time.

See the extra debugging output at https://travis-ci.org/baffo32/OpenBCI_NodeJS/jobs/171760737#L338 : the system receives many samples (instances of "a0" in the debug bytes), but processes only two of them ("sample 1", "sample 2").

aj-ptw, 2016-11-01:
This may be the same error as #119

baffo32, 2016-11-03:
After reviewing the Daisy behavior as determined by the firmware source code I've found that my perceptions here were wrong.  The samples are all processed correctly.  I'm sorry for reporting this as an issue of dropped samples.  It is probably actually an issue of being able to meet the requested samplerate under the extra overhead of the new promise instrumentation.

PR #117 (2016-10-31): Fix promise data processing
Ideally closes #115 by removing promises from all buffer processing functions.
closed 2016-11-03 by baffo32