openbci-archive/OpenBCI_NodeJS #146: Turn LSL example into a command line executable python script

teonbrooks, 2017-04-17:
@gabrielibagon wrote a pretty cool [example](https://github.com/OpenBCI/OpenBCI_NodeJS/blob/master/examples/labstreaminglayer/readme.md). we should simplify it to a simple command line executable so we can do something like this:
```
npm start openbci
python openbci_lsl
```
and everything just works

teonbrooks, 2017-04-17:
cc: @aj-ptw, @rodrigohubner

I can help out on getting the python part set up. just wanted to put it out there. will come back to this issue soon

aj-ptw, 2017-04-17:
All you need to do is spawn the Python and Node scripts in their own processes, so could be really easy to just have a super script that spawned both children

teonbrooks, 2019-04-10:
closing because this is no longer a priority. the example above is adequate

closed 2019-04-10 by teonbrooks