openbci-archive/OpenBCI_NodeJS #152: BUG: forgot parentheses in getBoardType() within the Cyton.prototype._finalizeNewSample function

nateGeorge, 2017-07-19:
I was trying to run the [get streaming daisy example](https://github.com/OpenBCI/OpenBCI_NodeJS/blob/master/examples/getStreamingDaisy/getStreamingDaisy.js), but it was crashing.  I found out just the parentheses were missing on the getBoardType() function, which was the source of the error.  For reference, the error looked like:

```bash
auto found board
using real board /dev/ttyUSB0
Serial port connected
Serial port open
Sending stop command, in case the device was left streaming...
Sending soft reset
Waiting for '$$$'
Channel 1: -0.00398862 Volts.
Channel 2: -0.00379756 Volts.
Channel 3: -0.00383619 Volts.
Channel 4: -0.00384175 Volts.
Channel 5: -0.07248861 Volts.
Channel 6: -0.07128130 Volts.
Channel 7: -0.07700699 Volts.
Channel 8: -0.08193068 Volts.
TypeError: Cannot read property 'toFixed' of undefined
    at Cyton.ourBoard.on (/home/nate/github/Regis-BCI-Research/OpenBCI_chrome_app_GUI/node_js_test2.js:31:71)
    at emitOne (events.js:96:13)
    at Cyton.emit (events.js:188:7)
    at Cyton._finalizeNewSample (/home/nate/github/Regis-BCI-Research/OpenBCI_chrome_app_GUI/node_modules/openbci/openBCICyton.js:2080:12)
    at _.forEach (/home/nate/github/Regis-BCI-Research/OpenBCI_chrome_app_GUI/node_modules/openbci/openBCICyton.js:1846:10)
    at arrayEach (/home/nate/github/Regis-BCI-Research/OpenBCI_chrome_app_GUI/node_modules/lodash/lodash.js:537:11)
    at Function.forEach (/home/nate/github/Regis-BCI-Research/OpenBCI_chrome_app_GUI/node_modules/lodash/lodash.js:9359:14)
    at Cyton._processDataBuffer (/home/nate/github/Regis-BCI-Research/OpenBCI_chrome_app_GUI/node_modules/openbci/openBCICyton.js:1839:5)
    at Cyton._processBytes (/home/nate/github/Regis-BCI-Research/OpenBCI_chrome_app_GUI/node_modules/openbci/openBCICyton.js:1813:26)
    at SerialPort.serial.on.data (/home/nate/github/Regis-BCI-Research/OpenBCI_chrome_app_GUI/node_modules/openbci/openBCICyton.js:292:12)
exit
stop streaming
Serial Port Closed
clean
```

Fixed it, and submitting a pull request.  By the way, it might be a good idea to use prefixes in commits like other open source software:
https://pandas.pydata.org/pandas-docs/stable/contributing.html#committing-your-code
https://udacity.github.io/git-styleguide/
Just a thought.

aj-ptw, 2017-07-19:
how did this issue get closed without referencing the issue number. Also i think you're right, i should start using those prefixes.

closed 2017-07-19 by aj-ptw

aj-ptw, 2017-07-20:
@nateGeorge i release your patch, thanks so much, my apologies. auto tests should catch that, looking into why they didn't. 

Also please checkout the commit summaries on the main page, i think you will like them! ;)

nateGeorge, 2017-07-20:
Sounds good!

PR #153 (2017-07-19): BUG: forgot parenthesis on getBoardType() func
Fixes #152