openbci-archive/OpenBCI_NodeJS #156: _count is undefined in sample

nateGeorge, 2017-07-20:
Trying to run the timesync examples, I've found that `_count` in `sample`, `sample._count` is undefined.  I think what this can be replaced with is sampleNumber, but it doesn't exactly work the same.  It's in this file: https://github.com/OpenBCI/OpenBCI_NodeJS/blob/2.0.0/openBCICyton.js
and the only place I see it is
`sampleObject._count = this.sampleCount++;`

The other problem I noticed is that the 'timestamp' property of 'sample' is always null, even after syncing.  Is this supposed to take some value other than null after syncing?

aj-ptw, 2017-07-20:
I just fixed the timestamp issue, I thought, are you on latest? I'll check on _count, count is supposed to be a running counter of how many samples you have received and it should not be replaced by sample nunber

nateGeorge, 2017-07-20:
Whoops, I was on 2.1.0, just upgraded to 2.1.2 and the timestamp is working.  When I console.log the sample, it looks like this:

```node
{ channelData: 
   [ -0.18750002235174446,
     -0.18750002235174446,
     -0.18750002235174446,
     -0.18750002235174446,
     -0.18750002235174446,
     -0.18750002235174446,
     -0.18750002235174446,
     -0.18750002235174446,
     -0.18750002235174446,
     -0.18750002235174446,
     -0.18750002235174446,
     -0.18750002235174446,
     -0.18750002235174446,
     -0.18750002235174446,
     -0.18750002235174446,
     -0.18750002235174446 ],
  sampleNumber: 33,
  auxData: { lower: <Buffer 00 00>, upper: <Buffer 00 00> },
  timestamp: 85700,
  _timestamps: { lower: 85698, upper: 85702 },
  valid: true }
```

No ._count in there.

aj-ptw, 2017-07-20:
Yup it's because I don't have an auto test to make sure count is there.... Adding and will probs release a new version later with the patch! You could put a counter on your 'sample' event counter to do the same thing In the meantime

nateGeorge, 2017-07-20:
Ok, thanks!  For now I'll just sync once every sample cycle.  Also just noticed that the accelData is sometimes there, sometimes not.  It would be nice to have it always there, and just keep the last value until it's updated.

aj-ptw, 2017-07-20:
Cool! Let me know if you have any problems. 

I'll think about it but I don't like the idea of adding static info to the sample object. Time syncing is different though because that accel packet comes in over multiple packets. With normal accel mode there is data every packet but most are zeros so you have to omit the packets with zeros in them. There is an internal object that stores the accel array as it builds, we could put a public accessor on it and you could read that every sample and forget about looking for it in each sample. Thoughts?

nateGeorge, 2017-07-20:
Hmm.  Maybe just leave it the way it is for now then and let people handle it as they see fit.  I guess adding a public method to access it wouldn't be a bad thing though.