openbci-archive/OpenBCI_NodeJS #157: Cyton sample rate is 125 Hz, firmware v2

nateGeorge, 2017-07-20:
Maybe this is just my setup, but so far when I do `ourBoard.sampleRate()` it returns 125.  I'm on v2 firmware, because when I call `ourBoard.usingVersionTwoFirmware()` it returns `true`.  The other strange thing is when I save the data, the sample numbers (`sample.sampleNumber`) range from 0 to 126 -- implying a sample rate of 127 Hz.  Sometimes the sample numbers start at 0, other times 1.  Sometimes the sample numbers end at 126, other times 127.  A sample data file is attached here: [test.csv.zip](https://github.com/OpenBCI/OpenBCI_NodeJS/files/1163936/test.csv.zip).  The timesteps are 0.008s gaps, but 1/127 is 0.00787, so it could just be rounding up.  Maybe we need to use the [utilities lib](https://github.com/OpenBCI/OpenBCI_NodeJS_Utilities) to set the sample rate to 250 upon connecting to a Cyton board.

I also noticed that when collecting data from the processing GUI, the timestamps are not uniform -- sometimes they are separated by 0.001 s, and other times 0.008.  I had assumed the default sample rate was 250 Hz, and had resampled the data to match 0.004 s timesteps, because the sampleNumber in the saved data files went up to 250.  But now I'm thinking the times that were saved from the processing GUI are times from the computer, not the board, and should basically be ignored.

aj-ptw, 2017-07-20:
Sample rate for Daisy over bluetooth is 125Hz. The gaps you are seeing are because of the serial port in the processing GUI. Lower your latency on your serial port. The nodejs will chop the upper sample number in half i believe, it's in the utility package.

In the test file why are there so many of the same sampleNumbers? Are you removing the `sample` event when you disconnect?

closed 2017-07-20 by aj-ptw

nateGeorge, 2017-07-21:
Yeah, I realized I had the .write function for the csv in a loop going through each channel, so each line was duplicated 16 times, haha.  I fixed it and replaced the zip file above.

So is 125 the max for Daisy over BT due to 16 channels?

nateGeorge, 2017-08-24:
So I guess there's no way to set the sample rate?  Is it possible to change the sample rate with either this lib or the utils lib?  When I collected data using the processing GUI, it appeared the sampling was at 250Hz, but with this lib it seems to be at 125 Hz.  What I've [read](https://github.com/WinslowStrong/OpenBCI_Wired_USB) is that the daisy sampling flip-flops, with the first 1-8 channels updating and then 0.004s later, the next 9-16 channels updating.  

Is it possible to set the sample rate with another API, like the [pre_activate](https://github.com/OpenBCI/OpenBCI_Python/blob/661552e31122833f3eca80c9d191bc4f40d55c08/plugin_interface.py#L25) function in Python?

aj-ptw, 2017-08-24:
There is no sample rate changing over Bluetooth/Serial.

125Hz Cyton with Daisy
250Hz Cyton

New wifi shield allows sample rate changing. On sale in a couple of weeks.

nateGeorge, 2017-08-24:
Ok, sounds good, thanks, looking forward to it.