openbci-archive/OpenBCI_NodeJS #162: Impedance for daisy issue

agneg, 2018-01-03:
Hello,

there seems to be an issue with impedance test for daisy module because this.options.daisy seems to be undefined.

openBCICyton.js line ~1389

Is that only me who has this issue?

Sincerely,
Agne

aj-ptw, 2018-01-03:
Please use [OpenBCI_NodeJS_Cyton](https://github.com/OpenBCI/OpenBCI_NodeJS_Cyton)

agneg, 2018-01-03:
Hmm, ok, may I ask why?

aj-ptw, 2018-01-03:
[Here is a blog post](http://openbci.com/community/nodejs-ecosystem-overhaul-cyton-stable-and-ganglion-in-beta/) explaining why.

agneg, 2018-01-03:
Ok, thank you for the link. That should probably be reflected in the README.

agneg, 2018-01-03:
Still the same issue though.

aj-ptw, 2018-01-03:
Okay, would you mind opening the issue in that repo with the proper line and steps to reproduce? 

Thanks!

agneg, 2018-01-04:
It is still exactly the same situation and function, line number slightly different. Just have any daisy example with impedance test and you will get 8 channels instead of 16. But nevermind, found the bug and fixed it myself.

aj-ptw, 2018-01-04:
@agneg can you please submit a pull request [to the repository](https://github.com/OpenBCI/OpenBCI_NodeJS_Cyton) with the bug fix? Would love to fix this for the rest of the community!

aj-ptw, 2018-01-04:
I'm really happy to hear you got the impedance testing working!

aj-ptw, 2018-01-04:
I merged in the development branch into master. This will show that the code for cyton does not live in this repo anymore! 

If you are not sure how to submit a pull request, I'm happy to teach you how and support your professional development!

Once again thanks so much for the bug report and sorry you found a bug in the code!

aj-ptw, 2018-01-04:
@agneg I'm going to close this issue because i opened issue [6 in Cyton Repo](https://github.com/OpenBCI/OpenBCI_NodeJS_Cyton/issues/6), please please share how you fixed the bug in that issue.

Please keep in mind this is an open source project and one of the best ways it gets better is when awesome people like you help solve bugs like this.

closed 2018-01-04 by aj-ptw

agneg, 2018-01-05:
@aj-ptw I will see if I can find a way to share it but quite frankly I keep unrolling more and more very strange behaviours in the code.

aj-ptw, 2018-01-05:
Thanks so much for your help! Please make sure your hardware firmware is up to date, that could for sure lead to problems.

Please be sure to open an issue for each strange behavior you find in the Cyton node js repo. The OpenBCI GUI uses the NodeJS drivers so we know they work really well for most things. 

If there is a function, such as impedance, that is not a core function of the repo, then maybe there is a bug. Maybe there is a setup issue in your code and I can tell you where the problem is. I would start with the get streaming examples and go from there.

I’m happy to help if you open new issues for each bug or problem you run into. Good luck!