openbci-archive/OpenBCI_NodeJS #165: Unable to resolve module `fs` from [Location]: Module does not exist in the module map

NaseebullahSafi, 2018-01-22:
I'm trying to use OpenBCI Cyton in react native application. However the issue is that I can't import it in react native. Is there a way to use Cyton in react native?

This is the error I get:
```
Loading dependency graph, done.
error: bundling failed: Error: Unable to resolve module `fs` from `C:\King's_College_London\Final_Year_Project\HeartFailure\node_modules\serialport\lib\serialport.js`: Module does not exist in the module map
```

and I try to use it like so:

```js
import Cyton from 'openbci-cyton';

  const ourBoard = new Cyton();
  ourBoard.connect("COM3").then(() => {
    ourBoard.on('ready', () => {
      ourBoard.streamStart();
      ourBoard.on('sample', (sample) => {
        for(let i = 0; i < ourBoard.numberOfChannels(); i++) {
          console.log(
            "Channel" + (i + 1) + ": " + sample.channelData[i].toFixed(8) + " Volts."
          );
        }
      });
    });
  });
```

aj-ptw, 2018-01-22:
Could this [openbci-rx](https://github.com/alexcastillo/openbci-observable) help?

aj-ptw, 2018-01-22:
Does node seriaport support react native?

NaseebullahSafi, 2018-01-22:
@aj-ptw aparently  RN doesn't support serialport

aj-ptw, 2018-01-22:
Checkout electron, or the openbci_hub

NaseebullahSafi, 2018-01-22:
Are you suggesting to use electron with react native? I have already played with Openbci_hub but only on web-based. However, my target is actually mobile.

baffo32, 2018-01-22:
@NaseebullahSafi It should be possible to use rn-nodeify and a replacement module such as react-native-usbserial to do this, but you would be treading new ground attempting it, and would likely run into hitches that might need you to look inside things and develop your own workarounds.

NaseebullahSafi, 2018-01-23:
Ok Thanks to @baffo32 and @aj-ptw I'm finally able to use the node module OpenBCI in react native. But the issue still remains; in that when OpenBCI module is imported:
```js
const Cyton = require('openbci-cyton');
```
We get "Maximum call stack size exceeded" error and so I presume that OpenBCI has a node event loop for I/O operations.

Is there anyway way we can fix this?

```
RangeError: C:/King's_College_London/Final_Year_Project/HeartFailure/node_modules/openbci-cyton/openBCICyton.js: Maximum call stack size exceeded
    at validate (C:\King's_College_London\Final_Year_Project\HeartFailure\node_modules\babel-types\lib\definitions\index.js:85:20)
    at validator (C:\King's_College_London\Final_Year_Project\HeartFailure\node_modules\babel-types\lib\definitions\index.js:57:7)
    at Object.validate (C:\King's_College_London\Final_Year_Project\HeartFailure\node_modules\babel-types\lib\definitions\index.js:190:10)
    at validate (C:\King's_College_London\Final_Year_Project\HeartFailure\node_modules\babel-types\lib\index.js:505:9)
    at Object.builder (C:\King's_College_London\Final_Year_Project\HeartFailure\node_modules\babel-types\lib\index.js:466:7)
    at PluginPass.CallExpression (C:\King's_College_London\Final_Year_Project\HeartFailure\node_modules\babel-plugin-rewrite-require\index.js:85:15)
    at newFn (C:\King's_College_London\Final_Year_Project\HeartFailure\node_modules\babel-traverse\lib\visitors.js:276:21)
    at NodePath._call (C:\King's_College_London\Final_Year_Project\HeartFailure\node_modules\babel-traverse\lib\path\context.js:76:18)
    at NodePath.call (C:\King's_College_London\Final_Year_Project\HeartFailure\node_modules\babel-traverse\lib\path\context.js:48:17)

```

xloem, 2018-01-23:
I don't know how to solve that issue, but just to add resources another alternative is the nodejs-mobile project, which takes the opposite approach of providing a full node runtime on mobile as a background process.  No clue if anybody's tried to make serial work with it.

NaseebullahSafi, 2018-01-23:
@xloem ok thanks. I'm just about to create a new project for testing with node-js in react native and try to make it work. I'll update you guys. But remember guys, this is a very serious limitation to OpenBCI-Cyton because many people are planning to create mobile apps for cyton rather than a web-based primarily because of portability or whatnot.

NaseebullahSafi, 2018-01-24:
@xloem I had a very long chat with the author of that project and although we were able to initialise openbci module in react, we were not able to use the serialport of nodejs.

@aj-ptw I would really appreciate if you could enlighten me. Is there any approach for simply connecting openbci to react native apart from nodejs?

aj-ptw, 2018-01-24:
If you want to start developing today, buy the WiFi Shield. You could already be streaming data! The OpenBCI Cyton is in need of a physical hardware update before this would be possible.

NaseebullahSafi, 2018-01-24:
Apologies if I have misunderstood you. But shouldn't it be possible for streaming data from the cyton to react native as long as the phone is connected to the PC along with the cyton dongle? @aj-ptw

aj-ptw, 2018-01-24:
I'm sorry I'm not really sure what react native is I guess. Can you explain to me where react native is running

NaseebullahSafi, 2018-01-24:
So react-native is Facebook's framework for cross platform mobile apps, targeting android and ios. It doesnt use node modules, which is why it's so difficult to use openbci-cyton node module in react-native. I have been searching for a solution for this but to no avail. I think the best choice would be then to use wi-fi sheild. Am I right? @aj-ptw

aj-ptw, 2018-01-24:
If you want to go direct to mobile phones, you can use the utilities module that's written in pure java-script to parse the raw data over wifi or use any of the existing browser compliant libraries available.

If you want to put a computer on you local network and have that send data to the mobile phone via TCP or something, you could plug the cyton dongle into the PC and stream to the mobile phone. Mobile phone don't have usb plug ports :(

octopicorn, 2018-01-24:
Hey @naseebullahsafi I speak React Native, hell, it’s all do all day long. How and(why) is RN the one thing you thought would connect to a SERIAL port? How do those things even meet? Is there a SERIAL poert on you phone?

(Hey AJ) this repo got real all of a sudden :)

NaseebullahSafi, 2018-01-24:
Correct indeed. So my question now would be, If I buy wi-fi shield for this cytone, would I be able to transmit data? If so is it also based on nodejs? Or a general wi-fi module that we can use to listen to?

octopicorn, 2018-01-24:
Is this one of those classic problems where there is a missing ‘fs’ module? We can’t take nodejs for granted. Fs = fileststem npm package . RN doesn’t get access to the FS because it doesn’t run, it’s compiled into totally deterministic code.

octopicorn, 2018-01-24:
From a shell on your machine you can run a nodejs process that will receive data on localhost. It sounds like (RN) you’d want the mobile phone to be the client?

octopicorn, 2018-01-24:
Yes WiFi shield transmits data. It is a server, WiFi receiver on your machine is client, which pipes to command line. So if you can pipe WiFi signal to the process that pings the SERIAL port you should be good.

BUT, how much streaming data do you plan to get through a SERIAL connection? Isn’t that speed measured in mbps? Serial is brutal, isn’t that like 16 steel pins?

octopicorn, 2018-01-24:
Imreading your earlier comment about “node event loop” and trying to understand how that fits in with the React component lifecycle, I think node just runs RN as a compiled bundle, they don’t communicate except maybe through a virtual connection, is that the SERIAL connection?

NaseebullahSafi, 2018-01-24:
Yes indeed @octopicorn. It's really pain in the ass to send data from the cytone dongle to react native. Simply put if we can't use nodejs modules in react native, we can't transmit data. But before going on and buying wi-fi shield, I wanted to make sure if it is really possible to transmit data to phone via WiFi shield. And find some implementations of how they do it.

NaseebullahSafi, 2018-01-24:
@octopicorn so the data should be streaming continuously until the user stops or closes the app. I'm not entirely sure how much data should be sent per second, but as long as some data is sent to phone, it would be good

octopicorn, 2018-01-24:
How about this? Is this related?
https://www.npmjs.com/package/serialport

octopicorn, 2018-01-24:
So I ask you this: what do you have your mobile that can connect to WiFi to stream in to your RN app? Regardless of what transmits on any protocol? What’s your receiver?

NaseebullahSafi, 2018-01-24:
@octopicorn ah yes, that was a wrong call. The problem was not with event loop, but rather importing the serial port in react native. But since serial port is node module it wouldn't compile and the app would crash

octopicorn, 2018-01-24:
I’m assuming SERIAL is just a port number you have mapped to a nodejs listener process?

NaseebullahSafi, 2018-01-24:
So currently I have a phone, and a cytone board with the dongle, and that's pretty much it really. I connect my cytone to PC via the dongle and tried to use OpenBCI module in react native, but wouldn't work

NaseebullahSafi, 2018-01-24:
@octopicornYes correct. But beware, we can't use serialport in react native, as its nodejs module. And we can't use react-native-serialport either because OpenBCI module uses nodes serialport

octopicorn, 2018-01-24:
@NaseebullahSafi 
I don’t know how much better to help
But want to communicate this to you: when you see ‘fs’ error like above it means your non module wants to access filesystem and it can’t because it doesn’t even have the base npm module ‘fs’ to do so.

99% if the time i see this, it means: I’m trying to use an npm idea that was meant from the core torun only with ‘fs’ access as a dependency.

NaseebullahSafi, 2018-01-24:
@octopicorn Really appreciate your time helping me understand some misconception. And after working on it for 3 days straight I now understand the reason behind it, but sadly there's literally nothing I can do to make it work.

aj-ptw, 2018-01-24:
How many channels do you want? There is a 2 channel 125Hz BLE driver that I wrote for Cyton that is still alpha but would allow you to at least get data from the Cyton. Although you would have to reprogram the physical RFDuino on the Cyton and would not be able to use the GUI because the dongle would no longer work. WiFi can definitely go to any web browser, which means your react mobile app can talk to the device!

NaseebullahSafi, 2018-01-24:
I see. Tbh Id just need 2 or 1 channel. What kind of changes would there be in rfduino? @aj-ptw

NaseebullahSafi, 2018-01-24:
You see,  willing to opt out from GUI and the dongle because the main priority is to create data transmission between cytone to react native.

NaseebullahSafi, 2018-01-24:
Since  doing an app for ekg/ecg, I'd only need 1 or 2 channel instead of all 8 .

aj-ptw, 2018-01-24:
Okay and you are able to use BLE inside react native?

NaseebullahSafi, 2018-01-24:
Yes I'm pretty sure:

https://github.com/innoveit/react-native-ble-manager

@aj-ptw

NaseebullahSafi, 2018-01-24:
Is there any possibility for the OpenBCI community to use the BLE for only 2 channels? As I fit in that category, and would really appreciate to use it.

aj-ptw, 2018-01-24:
The instructions and drivers are linked in the bottom. 

http://openbci.com/community/cyton-ble-code-now-in-alpha/

aj-ptw, 2018-01-24:
You could also get a ganglion instead of the cyton

NaseebullahSafi, 2018-01-24:
Ok thank you, I'll try to apply the same changes you made to `BoardWithBLE.ino` to make my cyton look for BLE

NaseebullahSafi, 2018-01-24:
@aj-ptw Sorry new to this Arduino world. In your post, you mentioned, "the RFduino firmware must import RFduinoBLE.h instead of RFduinoGZLL.h.". How can I do this?

I'm kind of confused what to change, to be honest.

NaseebullahSafi, 2018-01-24:
@aj-ptw I see you have this repo

https://github.com/PushTheWorld/OpenBCI_NodeJS_Cyton_BLE

Cant I just use this? I'm confused what changes do I have to do in my side of Cyton arduino.